b-it-bots/mas_domestic_robotics #6: Create a domestic ontology/integrate an existing one

argenos, 2018-03-07:
Worth 10 points.

We need an ontology that at least:

models semantic object knowledge (e.g. geometric object properties, contexts in which the object could be used, etc.)
associates geometric points to semantic locations

alex-mitrevski, 2018-12-23:
The groundwork for this has been laid with PR https://github.com/b-it-bots/mas_domestic_robotics/pull/63; the integration of the ontology with the rest of the code base is ongoing work, but the issue doesn't need to remain open at this point.

closed 2018-12-23 by alex-mitrevski

alex-mitrevski, 2018-12-25:
Just for the record, I've moved the ontology and the query interface (as well as the knowledge base interface discussed in issue https://github.com/b-it-bots/mas_domestic_robotics/issues/86) to a new repository: https://github.com/b-it-bots/mas_knowledge_base

There are two main reasons for this:
* This is already a relatively large piece of code and is better of independent since it could also be reusable by other applications
* Dependency management: `mas_execution_manager` and `mas_domestic_robotics` both depend on the knowledge base, so I didn't want to have a circular dependency between the two; this way, both of them depend on a separate repository