b-it-bots/mas_domestic_robotics #27: Switching path planners on the fly

argenos, 2018-04-09:
For some tasks we want to use a different path planner than our default (e.g. follow person with the carrot planner). The task is to create a node that can receive a `move_base` goal and pass it to a planner of choice. There are a few hints on how to go about it [here](https://answers.ros.org/question/265958/how-to-switch-the-planner-of-move_base-during-the-navigation/).

jcmayoral, 2018-08-13:
Hi.  I  thinkg it also makes sense to give a try to the move_base_flex (https://github.com/magazino/move_base_flex), since it also provides already a functionality to select the planners.

argenos, 2018-08-13:
Can they be switched online?

jcmayoral, 2018-08-13:
Yes. But it might require additional changes on the navigation actions. However, I guess it is smarter to provide the functionality on the move_base from now. I will do both, on some point. 

But one question, to change the planners online, what is required? A GUI? a service? a profile-based?

jcmayoral, 2018-09-01:
I think it makes sense to add this feature on mas_navigation repository https://github.com/b-it-bots/mas_navigation/pull/6

alex-mitrevski, 2018-09-01:
I agree, this is a general navigation functionality that belongs in `mas_navigation`.

closed 2019-02-15 by argenos