b-it-bots/mas_domestic_robotics #54: Saving map poses using spoken commands

alex-mitrevski, 2018-06-11:
To save predefined poses in a new map, we are currently using the `save_base_map_poses_to_file` script from the [ `mcr_navigation_tools`](https://github.com/b-it-bots/mas_navigation/tree/kinetic/mcr_navigation_tools) package. The process of saving poses using this script is however quite slow and tedious, as we have to drive our robot to the pose and then type in the name of the pose via the command line, which usually requires at least two people to walk around (one to drive the robot and another one to carry a keyboard and the robot's e-stop).

It would be considerably better to use spoken commands for saving these poses. This would allow us to drive the robot to a desired pose and then simply say "Save pose X" or something similar rather than typing the pose name; in addition, this would make it easier for a single person to record poses.

jcmayoral, 2018-06-13:
It seems interesting. Can I get this?

alex-mitrevski, 2018-08-03:
Resolved with pull request https://github.com/b-it-bots/mas_domestic_robotics/pull/55; closing the issue

closed 2018-08-03 by alex-mitrevski