b-it-bots/mas_domestic_robotics #58: Offline speech recognition

argenos, 2018-07-06:
Find an alternative to julius. 
Possible alternative: Pocket Sphinx

oarriaga, 2018-07-06:
For speech to text offline recognition I have tried the deep speech from mozilla and it worked fairly well; however, some post-processing would be required since it's predicting letters and not by words:

https://github.com/mozilla/DeepSpeech

jcmayoral, 2018-08-13:
I found this... I do not know if somebody has already experience with it https://github.com/cmusphinx/ros-pocketsphinx/tree/kinetic-devel

alex-mitrevski, 2018-08-13:
Yeah, I think Patrick has already tried using pocketsphinx.

argenos, 2018-08-13:
I've assigned @PatrickNa to mark it as "in progress", is there something we can start checking out as a pull request?

PatrickNa, 2018-08-13:
I've implemented Pocketsphinx and tested it on my laptop. The results were pretty bad and nothing comparable to what we have seen in Magdeburg last time. Nevertheless I will refactor the code and test it on the Robot. Probably in the end of the week I can make a solid pull request.

PatrickNa, 2018-08-13:
@robertocaiwu and I were already discussing on how to improve the model of pocketsphinx... However it was more brainstorming rather than concrete actions. We will intensify it in the next couple of weeks since it also effects RoboLand.

PatrickNa, 2018-10-20:
I would say this can be closed or at least rephrased. We have offline speech recognition now, but it can be further improved.

argenos, 2018-10-20:
@PatrickNa which pull request added it? Can you mention it here?

PatrickNa, 2018-10-20:
Offline speech recognition is possible since bd730722c9e958deaed6c54247609ba60f7ccf3c. 
Specifically: PocketSphinx is used when no internet connection is available. Have a look at 
mas_domestic_robotics/mdr_speech/mdr_speech_recognition/ros/scripts/speech_recognizer for further details.

argenos, 2018-10-20:
Good, we can close it. Not sure if I missed the documentation on how to setup pocketsphinx? I think we could open a new issue for that.

closed 2018-10-20 by argenos

PatrickNa, 2018-10-20:
Yes, we should continue open issues regarding speech.