b-it-bots/mas_domestic_robotics #66: add requirements.txt to Python packages

minhnh, 2018-08-25:
I propose adding `requirements.txt` to Python packages to handle Python dependencies. This will make maintaining dependencies modular to each package, which is easier than maintaining for the whole repository. We can start adding this to new packages and any packages with open pull requests.

argenos, 2018-08-25:
What would be an example of a package with python dependencies that cannot be resolved with rosdep? 

My other concern would be that we would need a script to loop through all *.txt files for the CI if we do this package-wise.

argenos, 2018-08-30:
I think rosdep actually supports pip. From their [documentation](http://docs.ros.org/independent/api/rosdep/html/commands.html#options):

```

  --as-root=INSTALLER_KEY:<bool>
                        Override whether sudo is used for a specific
                        installer, e.g. '--as-root pip:false' or '--as-root
                        "pip:no homebrew:yes"'. Can be specified multiple
                        times.

```

We should obviously test this, are you up for it?

minhnh, 2018-08-30:
yeah I can look into it

PatrickNa, 2018-10-21:
What is the status on this? I think this could be a possible solution for the installation of all speech relevant modules (e.g. PocketSphinx).

argenos, 2018-10-21:
We need to try it out. Can you add the dependencies to the package.xml? I still believe the right way to do this is using rosdep

argenos, 2019-05-23:
I will close this for now.

closed 2019-05-23 by argenos