b-it-bots/mas_domestic_robotics #74: Store speech relevant information in a database.

PatrickNa, 2018-10-16:
Store questions, answers, commands etc. in a database rather than in text files.

## Context

A database would provide a more central and structured environment for storing questions and answers.

## Expected Behavior

The speech packages (nodes) and actions would both refer to the same database rather than load the questions, answers, commands etc. from text files. Further, a database would allow to store additional information. For example the context of a question. 

## Actual Behavior

Instead of querying for the information we need we are loading text files, render the content and then look for the needed information. 

## Possible Fix

Set up a database for speech.

## Context

I'm refactoring the speech actions and improve the speech packages. Both need the same information to function.

argenos, 2018-10-16:
In my opinion, these are things that should be either stored in the knowledge base or the ontology (see #6). I don't think we should be storing this in a different database and wouldn't really devote time to this until the KB + ontology is done, and while we have the text files.

argenos, 2018-10-16:
@alex-mitrevski may have something to add to this

PatrickNa, 2018-10-16:
As long as it is a database where we can maintain this information I'm fine with it. I did not know that there is already something. I would still leave this issue open until we have integrated a database usage into our speech packages.