b-it-bots/mas_domestic_robotics #78: Implement Kaldi as an additional offline recognition tool.

PatrickNa, 2018-10-20:
So far only PocketSphinx is used for offline speech recognition. The model of this tool is ok, but not sufficient for our use cases. Rather than creating an own model, another speech recognition tool and its model, are suppose to help out. The tool is Kaldi and has shown promising results in the RnD project of @robertocaiwu.

## Expected Behavior

Both PocketSphinx and Kaldi are suppose to analyze the vocal input and give out their results. The better result will be taken as what the user has said.

## Actual Behavior

Only PocketSphinx is implemented as offline speech recognition tool.

## Context

Improve the offline speech recognition results by adding a second tool.

alex-mitrevski, 2018-10-21:
If we have multiple models, how are we going to decide which one gives a "better" result? Given an unknown sentence, this seems like a very difficult task. I suppose we could somehow check the recognised sentences from both models against an ontology, see if they make sense, and then take the one that matches the ontology more, but then what if they both make sense from that point of view?

Wouldn't it be better to decide on using one of the two for offline speech recognition and stick to it (the one that gives better results on a test data set)?

PatrickNa, 2018-10-21:
Exactly, the check against the ontology is suppose to decide which approach gave us the better result. At the moment I cannot imagine that both will provide two completely different, but valid sentences. However if this would be the case we still have the distances (likelihoods). For the case that both would be the same we would chose one randomly or prefer on approach over the other. 

And yes you are right, we don't know yet if it makes sense to have two or just stick with one (in terms of efficiency and performance). I guess we can approach this problem scientifically and run some tests to get an answer to this question.

argenos, 2019-05-22:
This is implemented in #145 , correct?

closed 2019-05-23 by PatrickNa