b-it-bots/mas_domestic_robotics #79: Implement a weather API for the question "how is the weather".

PatrickNa, 2018-10-24:
The answer for the question "how is the weather" (or a modification of this question) should not be hard-coded. The information about how the weather is or how it is going to be, can be taken from a weather API.

## Context

Hard-coding the weather information is sufficient for demos, but it is not feasible for the long term. Since methods for getting the weather information exist, they also should be used. 

## Expected Behavior

The robot requests the weather information through an API from the internet and responds to the user accordingly. 

## Actual Behavior

The robot looks up what answer has been saved for the question and says it out loud. For example, even if the weather is bad and it is cold outside, it would tell the user that it is sunny.

## Possible Fix

Implement an API like for example the one from https://openweathermap.org/api.

## Steps to Reproduce

1. In the mdr_question_answering package check if the question is weather related.
2. If so, then make a request to your weather API.
3. Put the received information in a nice response.
4. Forward it to the output node to let the robot say it out loud.

## Context

A question about the weather is not static, by that I mean the answer to this question changes over days. The robot should therefore be able to adjust its answer to the given environmental conditions.

PatrickNa, 2018-10-26:
@henrikschnor Eventually you will find some good references here: https://www.programmableweb.com/news/top-10-weather-apis/analysis/2014/11/13

rameshjesswani, 2018-10-27:
@PatrickNa  actually rasa provides functionality to find the intents and entities from the sentence. And based on predicted intent, we can write the custom actions to get the data from weather api. Here is simple example: [rasa-weather-chatbot](https://medium.com/coinmonks/rasa-python-weather-chatbot-51fc218d346d)

PatrickNa, 2018-10-27:
@rameshjesswani Thanks for the example! I guess it is also interesting for @ErickKramer and @argenos , since they are working on a NLP related research project and are about to implement such a solution for the @home team, too. I see a general interest from your side regarding this project? Would you like to join an @home meeting to eventually discuss this topic further?

rameshjesswani, 2018-10-27:
@PatrickNa yes sure I can join on next meeting and we can discuss more about this :)

argenos, 2018-10-27:
@ErickKramer I think it's rasa-nlu is something definitely worth looking into

alex-mitrevski, 2018-12-03:
This was addressed by pull request https://github.com/b-it-bots/mas_domestic_robotics/pull/82, so I'm closing the issue.

closed 2018-12-03 by alex-mitrevski