b-it-bots/mas_domestic_robotics #83: Question classification and method allocation.

PatrickNa, 2018-11-06:
The issue is opened, because different methods to approach questions are in discussion. Thus it needs to be specified which methods are most suitable for a given type of question. For example the questions from RoboCup are matched to a certain answer, which can be hard-coded. On the other hand questions with dynamic answers exist. 

## Context

The implemented speech solution focuses on string matching approaches only. This seems to be a well suited approach for questions with a specific answer. However is a conversation with the robot, which is similar to a conversation with a human not possible, yet. Therefore natural language processing approaches (NLP) have been suggested. Before applying, we first need to discuss for which type of question and in which situation NLP is a valid approach.

## Expected Behavior

From this discussion a way to select one of the above mentioned methods is suppose to be emerged.

## Steps to Reproduce - Questions that should be answered

1. How to categorize questions?/ Do categories already exist?
2. How to distinguish between the categories?
3. What have other people done that could be of value for us?

PatrickNa, 2018-11-06:
@rameshjesswani

PatrickNa, 2018-11-06:
I opened this issue to discuss the information flow and the architecture for our speech component. However I would also like to start developing in parallel. Therefore I opened issue  #84, which is a prerequisite for implementing Rasa.

argenos, 2019-05-22:
Is this addressed in #145 ?

PatrickNa, 2019-05-23:
At this point I'm not sure how relevant this is, anymore. I will close it for now, since we are able to deal different use cases (see RoboCup scenarios).

closed 2019-05-23 by PatrickNa