b-it-bots/mas_domestic_robotics #84: ROS Wrapper for Rasa.

PatrickNa, 2018-11-06:
In order to integrate the chat bot functionality of Rasa into our robot, we need an interface for ROS. 

## Context

We want to implement an NLU approach and focus on the work of Rasa. Since the focus of this library is on chat bots, we need to provide a wrapper for ROS.

## Expected Behavior

Rasa can be used within our existing speech code.

## Steps to Reproduce - Those are general suggestions, that might change during the process.

1. Understand how Rasa works in general: What are the inputs and what are the outputs?
2. Understand the information flow in the robot and answer the following question: How to provide the needed inputs within the robot?

rameshjesswani, 2018-11-09:
I have created diagram that gives overview of rasa and its functionality. We can use trained model by sending http request to the model as shown here [HTTP API](https://rasa.com/docs/core/server/)

[image]

PatrickNa, 2018-11-09:
Great! That means that the ROS wrapper can be easily created. Looking forward to the meeting today!

argenos, 2019-05-22:
This was solved in #85 ?

PatrickNa, 2019-05-23:
I will check.

argenos, 2019-06-30:
I'll close this for now. We can reopen if it hasn't been solved.

closed 2019-06-30 by argenos