b-it-bots/mas_domestic_robotics #102: Implement a Noise Cancelling Solution.

PatrickNa, 2019-03-04:
This is a must have for the speech component. During the competition it will be noisy, therefore we need to have a solution to still be able to recognize the speech.

## Context
A pre-process for the speech component.

## Expected Behavior
The microphone input is filtered.

## Actual Behavior
Microphone input is only "filtered" by python speech recognition, but this is not very efficient.

henrikschnor, 2019-03-05:
I have a few pending changes for mdr_speech_recognition locally, where I got RNNoise (https://people.xiph.org/~jm/demo/rnnoise/) working. I tried combinations of this together with the voice activation detection algorithm from webrtc to improve reliability of speech input. It seems to be working quite well, however I could not test it properly yet and there are still a few things that need to be fixed/documented/improved.

I'll probably get back to it when I start implementing rasa for the "Serving Drinks" scenario.

argenos, 2019-05-23:
This can be closed, right?

PatrickNa, 2019-05-23:
In my opinion this is still an open issue. Current state: We are using Ubuntu's echo cancelling setting and analyze the recognized speech further. By doing this we are able to get the intention for of the user, but this holds true for restricted scenarios, only (e.g. "where is this"). If we would like to have a more open conversation we need to rely more on what has been recognized and therefore we need to have a more filtered input already.