b-it-bots/mas_domestic_robotics #103: Implement Rasa Core.

PatrickNa, 2019-03-04:
Rasa Core is foreseen for the processing of conversation. It needs to be implemented into the ROS architecture.

## Context
This component is essential for speech. It is suppose to allow conversation between the robot and humans and is considered as the central unit for managing the verbal and contextual information.

## Expected Behavior
Based on a conversation, certain actions need to be carried out. 

Rasa core needs to be implemented in such a way, that scenarios for the RoboCup competition can be easily implemented.

argenos, 2019-05-23:
Can you update or close this issue please?

PatrickNa, 2019-05-23:
Will do. Probably this afternoon.

argenos, 2019-05-25:
I'll close this for now. If need be, we can open a new issue.

closed 2019-05-25 by argenos