b-it-bots/mas_domestic_robotics #104: Establish a Strategy for Learning new Objects.

PatrickNa, 2019-03-04:
In general everybody knows how to train a new model for objects. However it is time to work out and automate this process. This includes the taking of pictures, labeling etc.

## Context
Object recognition is important in a lot of tasks. For example storing groceries.
Since the current object recognition model is limited, it is important to come up with a new one, so that scenarios can be properly tested. Additionally, the objective of this issue is important for the competition itself. This is due to the fact that new objects can be added just before the competition and we need to be able to react to the changes.

## Expected Behavior
Process is divided in well defined steps and every team member will know what to do.

## Actual Behavior
A vague idea exists of how it could work.

minhnh, 2019-03-05:
Currently, we use the Single-Shot Multibox Detector algorithm for object detection, which in general requires the following steps for training a new model (@ErickKramer please add anything I missed):
* Collect and label data: for each image, ground truth annotations of objects are needed
* Training: we need a common train script developed for the different detection models, right now only the one which comes with `ssd_keras` is available
* We need to automate the data augmentation process from the annotated boxes, i.e. essentially putting them onto different backgrounds. This has been done in Naresh's R&D, but we still need to integrate.

Object recognition/classification has been using built-in Keras image classification models, which are a bit simpler to train. @robertocaiwu please document somewhere what you've done during the SciRoc event for this task.

argenos, 2019-03-12:
Is there a script for training already? If not, I’d suggest opening an issue specifically for that in mas_perception and linking it here.

minhnh, 2019-03-12:
@ErickKramer was supposed to make the training script with the tutorial which may possibly use the [common data interface I created](https://github.com/minhnh/dataset_interface). I can create an issue and add him.

argenos, 2019-03-12:
Yeah, go ahead!

argenos, 2019-05-25:
@minhnh we can close this, right?

minhnh, 2019-05-25:
Yes. Documentation for training is available after b-it-bots/dataset_interface#1 is merged.

closed 2019-05-25 by argenos