b-it-bots/mas_domestic_robotics #106: Define Scenario "Serving Drinks".

PatrickNa, 2019-03-05:
We have prepared the scenario storing groceries already. Now it is time to proceed with the next task and set it up in parallel. The scenario suggested in this issue is the "Serving Drinks" scenario.

## Context
The "Serving Drinks" scenario is described in the newest rule book of the RoboCup competition (2019). 

## Expected Behavior
Similar to storing groceries a scenario in mdr_robocup_tasks needs to be defined.

henrikschnor, 2019-03-05:
Is this issue specifically for creating the scenario setup and the state machine? If not, could we edit this one? I created another issue about serving drinks (see above) where I added some of my thoughts.

PatrickNa, 2019-03-05:
Yes. It is specific for creating the scenario setup and the state machine.

henrikschnor, 2019-03-07:
There is a bonus reward available for this task:
> 1.  Preemptively informing a guest that the ordered drink is not available (250pts)

To do this, we need to go to the bar and observe it for available drinks before we take orders. Since we already have code for this, I'd say we try to do this (there is no penalty if we fail).

closed 2019-06-14 by argenos