b-it-bots/mas_domestic_robotics #107: Define Scenario "Bring out the garbage".

PatrickNa, 2019-03-05:
We have prepared the scenario storing groceries already. Now it is time to proceed with the next task and set it up in parallel. The scenario suggested in this issue is the "Bring out the garbage" scenario.

## Context
The "Bring out the garbage" scenario is described in the newest rule book of the RoboCup competition (2019). 

## Expected Behavior
Similar to storing groceries a scenario in mdr_robocup_tasks needs to be defined.

robertocaiwu, 2019-03-05:
You can assign this issue to me.

robertocaiwu, 2019-03-16:
Related to the **Take out the garbage** task [#110]:

- [ ] Component to turn head and /or body to a certain point [#111]
- Perception
    - [ ] Recognize if object is a bin [#112]
    - [ ] Recognize if there is a lid on the bin[#113]
    - [ ] Recognize if the bin is empty or not [#114]
    - [ ] Detect where to grasp the trash bag or other objects from. [#115]
- Manipulation
     - [ ] Grasp verification

argenos, 2019-05-08:
There is already some progress in https://github.com/b-it-bots/mas_domestic_robotics/commit/6d080225d9afeb48eb9d690d346e25ac58d3305e, but the final scenario tested in German Open is not pushed. @robertocaiwu can you make sure we get the latest changes from the robot?

argenos, 2019-05-23:
Please update or close this

robertocaiwu, 2019-05-23:
Duplicate issue [#110]

closed 2019-05-23 by robertocaiwu