b-it-bots/mas_domestic_robotics #111: Component to turn head to a certain point

robertocaiwu, 2019-03-05:
We need a new skill to move/turn the head towards a specific point in order to perceive.

# Context

This component is needed for the _Take out the garbage_ [see #110] task in order to find the trash bins located on the arena.

minhnh, 2019-03-05:
I would rename this to a skill which configures the whole robot to look at a point of interest, i.e. a `LookUpAction`. This skill may include:
* turn the head to an appropriate angle
* raise or lower the head to an appropriate height
* move the arm out of the way of the cameras

After the above operations, the skill can be included in a larger skill or extended to handle #44, which would also invovle:
* navigate to a general area which may have a subject of interest
* look around the area to find the subject
* find an ideal location to observe the subject

@argenos @alex-mitrevski this may need to be in a separated, bigger issue/project

argenos, 2019-03-12:
The first 3 bullets sound to me like a perfect use for FTSM. I agree the second part belongs on a different PR (and skill).

minhnh, 2019-03-21:
Here's a [helpful post](https://gamedev.stackexchange.com/questions/79172/checking-if-a-vector-is-contained-inside-a-viewing-frustum) on getting the distances from a 3D point to each plane of the camera's frustum. I believe the distances would be then in world coordinates (e.g. meters).