b-it-bots/mas_domestic_robotics #116: Student projects for SDP

argenos, 2019-03-06:
For next semester, it would be good if we can have some projects to propose to the SDP students. The projects should preferably
* have a "starting point", e.g. some existing code, a tutorial, some old package that you found somewhere else
* have a longer scope than SDP  

If you have any ideas that you think would fit, please open an issue for it, and add the link as a comment here.

argenos, 2019-03-06:
cc @sthoduka

argenos, 2019-03-06:
@rhasel4s once you open the issue with the planning actions, please link it here.

minhnh, 2019-03-06:
Depending on what's done in Koblenz, extension and improvement of b-it-bots/mas_perception#31 may be a fitting task.

alex-mitrevski, 2019-03-10:
This could be a small, but useful project: https://github.com/b-it-bots/mas_execution_manager/issues/9

argenos, 2019-03-15:
I've opened #126 to discuss the missing planning actions.