b-it-bots/mas_domestic_robotics #117: mdr_listen_action fails to build

henrikschnor, 2019-03-07:
The error message is as follows:
```
______________________________________________________________________________________________________________________________________________
Errors     << mdr_listen_action:install /home/henrik/ros/mas_stable/logs/mdr_listen_action/build.install.004.log                              
error: package directory 'common/src/mdr_listen_action' does not exist
CMake Error at catkin_generated/safe_execute_install.cmake:4 (message):
  
  execute_process(/home/henrik/ros/mas_stable/build/mdr_listen_action/catkin_generated/python_distutils_install.sh)
  returned error code
Call Stack (most recent call first):
  cmake_install.cmake:118 (include)
  

make: *** [install] Error 1
cd /home/henrik/ros/mas_stable/build/mdr_listen_action; catkin build --get-env mdr_listen_action | catkin env -si  /usr/bin/make install; cd -
..............................................................................................................................................
```

It seems like the folder `common/src/mdr_listen_action` is missing. @PatrickNa could you look into this?

PatrickNa, 2019-03-10:
I could not really reproduce this error... Does it still appear after you fixed issue #122?

alex-mitrevski, 2019-03-10:
I also can't reproduce the error, but since the package doesn't seem to have a `common` directory, it makes sense to update the `setup.py` so that it doesn't try to set up a non-existing package.

henrikschnor, 2019-03-11:
To reproduce the error, build the packages with Docker, just as Travis does:
```sh
git clone -b devel https://github.com/b-it-bots/mas_domestic_robotics.git
cd mas_domestic_robotics
docker build .
```
This takes some time but eventually it will (still) fail with the error I posted above.

The code expects some messages to be in `mdr_listen_action.msg`, while `mdr_listen_action` points to the nonexistent `common` directory. I'm not sure where the automatically generate message files from `Listen.action` go, but it seems like ros also generates a package for that.

Anyway, I tried building it with the reference to `mdr_listen_action` in `setup.py` removed and it seems to build and run just fine. If no one does, I'll create a PR tomorrow.

minhnh, 2019-03-12:
I think the issue is this [extra line here](https://github.com/b-it-bots/mas_domestic_robotics/blob/db9611c581f1dc5aff156ab60f224a3f7435e7e1/mdr_planning/mdr_actions/mdr_speech_actions/mdr_listen_action/setup.py#L8), which points to a non-existent folder. Should I add that to my PR, or should it be handled in a separate PR @argenos.

argenos, 2019-03-12:
Its own PR would be best!

closed 2019-03-14 by henrikschnor