b-it-bots/mas_domestic_robotics #122: Travis builds fail

henrikschnor, 2019-03-09:
Travis builds for mas_domestic_robotics fail for every opened pull request because some dependency fails to build.

moriarty, 2019-03-09:
https://travis-ci.org/b-it-bots/mas_domestic_robotics/builds/503674866#L1610

```
Starting >>> ae_test_scenario_visualiser                                       
Finished <<< ae_msgs                                            [ 5.9 seconds ]
Starting >>> ae_test_scenarios                                                 
_______________________________________________________________________________
Errors << ae_test_scenario_visualiser:install /kinetic/logs/ae_test_scenario_visualiser/build.install.000.log
error: package directory 'ros/src/ae_test_scenario_visualiser' does not exist
CMake Error at catkin_generated/safe_execute_install.cmake:4 (message):
  
  execute_process(/kinetic/build/ae_test_scenario_visualiser/catkin_generated/python_distutils_install.sh)
  returned error code
Call Stack (most recent call first):
  cmake_install.cmake:118 (include)
  
make: *** [install] Error 1
cd /kinetic/build/ae_test_scenario_visualiser; catkin build --get-env ae_test_scenario_visualiser | catkin env -si  /usr/bin/make install; cd -
...............................................................................
Failed << ae_test_scenario_visualiser:install                 [ Exited with code 2 ]
Failed <<< ae_test_scenario_visualiser                        [ 3.3 seconds ]  
```

henrikschnor, 2019-03-09:
Issue #117 needs to be fixed and #123 merged. With those changes Travis builds fine for me.

closed 2019-03-14 by henrikschnor