b-it-bots/mas_domestic_robotics #126: Planning actions for new robocup scenarios

argenos, 2019-03-15:
For some of the SDP projects, I would like students to develop missing skills or actions. Could you please add a list here of the things you have come up so far? Please add to which scenarios they belong in case we have too many, this will help with prioritization. 

Related to #105, #106, #107, #108, #110.

robertocaiwu, 2019-03-16:
Related to the **Take out the garbage** task [#110]:

- [ ] Component to turn head and /or body to a certain point [#111]
- Perception
    - [ ] Recognize if object is a bin [#112]
    - [ ] Recognize if there is a lid on the bin[#113]
    - [ ] Recognize if the bin is empty or not [#114]
    - [ ] Detect where to grasp the trash bag or other objects from. [#115]
- Manipulation
     - [] How to grasp a trash bag
     - [ ] Grasp verification

argenos, 2019-03-20:
@rhasel4s @henrikschnor @PatrickNa the deadline for this is Friday, has there been any progress? I still need to work on the proposals a bit after you guys suggest them.

henrikschnor, 2019-03-20:
I'm not sure which scope those SDPs are supposed to have, but here are some things that need to be done for "serving drinks":
- Detect all people in a room
- Detect faces of (detected) people and remember them (there might already be some code for the detection part [here](https://github.com/b-it-bots/mas_domestic_robotics/tree/devel/mdr_planning/mdr_scenarios/mdr_demos/mdr_demo_describe_people))
- Recognize what a person is holding in their hands
- Create an action/behaviour for picking up specific objects

rhasel4s, 2019-03-21:
Hey, besides the actions or skills we could need a JSHOP2 dispatcher for ROSPlan. At the moment we are using the standard dispatcher which has a slightly different plan representation. At the moment shop is modified to generate the plan in the same way the popf planner does.

Otherwise @henrikschnor @robertocaiwu and @PatrickNa have already defined the skills needed for the tasks. 

I think at one point we should discuss about what each skill is doing and the naming of the operators and predicates for the preconditions, add and delete actions they will add to the KB, these should also be defined so that the students stick to these conventions, this will save a lot of refactoring once they get integrated.

argenos, 2019-03-21:
> naming of the operators and predicates for the preconditions, add and delete actions they will add to the KB

Let's add that to this discussion thread, so I can refer back to it when writing the proposals. 
Could you maybe add those for the actions that are already listed?

rhasel4s, 2019-03-24:
| Operator | Precondition | Delete | Add |
| --- | --- | --- | ---  |
|move_base | (waypoint ?to) <br/>  |  (robot_at ?robot ?from)   |  (robot_at ?robot ?to)   |
| open_door | (robot_at ?robot ?waypoint) <br/> (door ?door) <br/> (door_status ?door closed) <br/>      |  (door_status ?door closed)     |  (door_status ?door open)     |
| perceive_plane | (robot_at ?robot ?waypoint) <br/>(belongs_to ?plane ?waypoint)<br/>(perceive ?plane ?waypoint) | (perceive ?plane ?waypoint) | (perceived ?plane ?waypoint) |
| pick    |(robot_at ?robot ?waypoint)<br/>(plane ?plane)<br/>(perceived ?plane ?waypoint)<br/>(belongs_to ?plane ?waypoint)<br/>(on_plane ?object ?plane) <br/>(store_object ?object)<br/>(emptyGripper ?robot)<br/>|(on_plane ?object ?plane)<br/>(store_object ?object)<br/>(emptyGripper ?robot)| (holding ?robot ?object)|
| place | (perceived ?plane ?waypoint)<br/>(belongs_to ?plane ?waypoint)<br/> (holding ?robot ?object) | (holding ?robot ?object) | (emptyGripper ?robot)<br/>(on_plane ?object ?plane) |
| locate  |       |       |       |
| explore |       |       |       |
| ...     |       |       |       |

i started with the once we allready use in the domain.

alex-mitrevski, 2019-03-24:
@rhasel4s The client of `perceive_plane` does more than what you have in the table: for each object, an `on(object, plane)` and `object_category(object, category)` predicates are added + the plane is marked as `explored`. Also, before perceiving, all objects on the plane are deleted from the KB. Implementation [here](https://github.com/b-it-bots/mas_domestic_robotics/blob/devel/mdr_planning/mdr_actions/mdr_perception_actions/mdr_perceive_plane_action/ros/scripts/perceive_plane_client#L60). The other actions match what we have.

rhasel4s, 2019-03-25:
ok, i add those things, we should then progress with the explore and locate behaviour. 

- explore - would be exploring something unknown like a room, piece of furniture or whatever.

- locate - would be identifying an object at specific location. 

so explore would mostly also include the locate action, but the locate action can be used when we  already know the location but not the exact position.

closed 2019-05-22 by argenos