b-it-bots/mas_domestic_robotics #140: README has broken link

ejichen, 2019-05-07:
## Context
https://github.com/b-it-bots/mas_domestic_robotics#getting-started

Has a link:

> ## Getting started
> 
> b-it-bots members can use [these instructions](https://github.com/b-it-bots/dev-env#setup) to setup a complete development environment for all our robots.
> 
> For external users, the following instructions should get you a working system:

The section does say "b-it-bots mebers" but I ( @moriarty ) still think that the file which is being linked to should be open as it helps anyone who might want to re-use this code.

## Expected Behavior
The link to work

## Actual Behavior
The link doesn't work

## Possible Fix
make that repo open source, or move that portion of the repo to the public somewhere else.

## Steps to Reproduce
1. Don't log into GitHub
2. Click the link

## Context
I ( @moriarty ) was giving some advice to go look at some of your code for an example.

## Your Environment
Any browser

moriarty, 2019-05-08:
Just confirming, I filed that ticket from someone else's computer.

I don't see anything in the dev-env repo which is worth keeping private, it's basically a newer version of 3 other repos, which it cites at the bottom of it's readme.

moriarty, 2019-05-08:
For Context, I've asked @ejichen to investigate two open-sourced approaches to solving a task.

So she was trying to get something from here running on her robot.

moriarty, 2019-05-08:
https://github.com/b-it-bots/dev-env/issues/12 this can be closed as a duplicate. I knew I ran into this issue before.

argenos, 2019-05-08:
I still haven’t had the chance to clean up that repo, which is why it’s still private. I’ll give access to @ejichen later today. 

FWIW the repo is unfortunately not updated with all the dependencies, so PRs are welcome if you run into problems.

ejichen, 2019-05-08:
Thank you for giving me access!

closed 2019-05-08 by argenos