b-it-bots/mas_domestic_robotics #141: Using the web-based socket for HRI

argenos, 2019-05-08:
Could we integrate the code we used for the screen during `where is this?` into this repository? Or can we get a fork from the `RoboLand` repo? I think that visual information on the robot is super useful, and we are currently wasting the screen.

robertocaiwu, 2019-05-14:
The one from roboland is very amy specific. I created a new repo on my git for the GO. I would need to clean it up and remove GO specific stuff. Also, we could develope it further and add more things.

Link to the repo: https://github.com/robertocaiwu/mas_web_interface

PatrickNa, 2019-05-14:
Yes. I will create a repository for that and we start with what we have developed in Magdeburg, but should clean it and improve it quickly.

Suggestion from Alex and me regarding the repository name: `mas_hri_interface`

argenos, 2019-05-14:
should we fork this one and start from there? or is it really better to start from scratch?

PatrickNa, 2019-05-14:
Can we fork, but rename it? If yes, then we should do it. Otherwise I would prefer the more general name for the repository. In that case I would ask Roberto to push to the newly created repo and we continue the work from there.

closed 2019-05-14 by PatrickNa