b-it-bots/mas_domestic_robotics #143: Testing find my mates

argenos, 2019-05-08:
This one doesn't really require further description, but @PatrickNa you can add any further things that remained as todos.

argenos, 2019-06-30:
@robertocaiwu could you help us out with running this test (maybe with the help of @henrikschnor if he has some available time during exam season?)?

The important thing out of this is finding out already things that don't work or need to be updated to work on the lab instead of GO.

robertocaiwu, 2019-07-15:
I met with @henrikschnor  and we identified a few problems when trying to run this scenario. Since issues/ pull request have been raised, I will link them:

- [x] Speech models was not pointing to the correct directory. Solved [#195]
- [ ] NLU Rasa models for find my mates is not present. (needed for "find" intend and for names and description of persons) [b-it-bots/mas_models/issues/5]
- [ ] Find people package was not running. (Tried running the package by itself, but does not seem to return a result.)  [#152]

alex-mitrevski, 2019-07-15:
@AhmedFaisal95 and I will need to work on the people finding action on Friday/Saturday; this is related to his R&D, but we'll tell you how it goes (and we'll fix any issues as we face them).

closed 2020-02-13 by alex-mitrevski