b-it-bots/mas_domestic_robotics #172: Clean up of mdr_find_people

argenos, 2019-05-25:
The current version of the package contains comments in the package.xml and CMakeLists.txt files. These files also need to be updated to handle the dependencies correctly.

argenos, 2019-05-25:
This topic should be passed as an argument https://github.com/b-it-bots/mas_domestic_robotics/blob/fb37f5591ce8d83c3f279f3bd9e052741aca7581/mdr_planning/mdr_actions/mdr_perception_actions/mdr_find_people/ros/src/mdr_find_people/action_states.py#L18

And we need an updated launcher for `mas_hsr` which then modifies that topic to the HSR-specific one.

alex-mitrevski, 2019-05-25:
One other thing that needs to be done is reimplementing the action using FTSM.

minhnh, 2019-05-25:
duplicate #152?

alex-mitrevski, 2019-05-25:
I don't think this is a duplicate. This issue is about small refactoring of the existing package; your issue is about the logic behind the action itself, which I see as considerably more involved.

alex-mitrevski, 2019-06-15:
The cleanup has already been done (commits https://github.com/b-it-bots/mas_domestic_robotics/commit/52e0c8eb2c54966d5dfa2f3f2eabf7bbc67bfc73, https://github.com/b-it-bots/mas_domestic_robotics/pull/190/commits/f4b9b8b61d63be0c60966572a30f75060d744e19, and https://github.com/b-it-bots/mas_domestic_robotics/pull/193/commits/9ea3c8e355160ec4731471e5049b1604a51c9c4b), so I'll close the issue. The migration to FTSM can/should be a separate issue.

closed 2019-06-15 by alex-mitrevski