b-it-bots/mas_domestic_robotics #174: Clean up of where is this

argenos, 2019-05-25:
The rasa models we are loading are right now hard coded, see [this](https://github.com/b-it-bots/mas_domestic_robotics/blob/devel/mdr_planning/mdr_scenarios/mdr_robocup_tasks/mdr_where_is_this/common/model/current/nlu/checkpoint) for an example. It probably makes sense to move the models to `mas_models` as well. 

@alex-mitrevski @minhnh thoughts?

minhnh, 2019-05-25:
There's a [utility function](https://github.com/b-it-bots/mas_perception_libs/blob/055a1e44984dc829fc4b1ea4e03b81eb072e32f7/ros/src/mas_perception_libs/utils.py#L54) that may make it easier to get files from packages' relative paths. We can put these type of files in the model repository as a ROS package and access them this way.

argenos, 2019-05-25:
Sounds good to me, can you open a PR for that?

minhnh, 2019-05-25:
Do you mean to upload a model? Shouldn't @ErickKramer handle this?

argenos, 2019-05-25:
I meant the utility function. Need a proposal on how to handle it, e.g. do we want to make `mas_models` dependent on `mas_perception_libs`?

alex-mitrevski, 2019-05-25:
I don't think being dependent on `mas_perception_libs` makes sense in that context. It would be better if such utility functions are stored elsewhere, perhaps in a dedicated repository, since they could be useful in different places (e.g. we need something similar in the execution manager for handling the state machine hierarchies in the code rather than through launch files).

minhnh, 2019-05-25:
I'm for a small utility repository for this kind small functionalities. We should try to keep it compact and as a single package though, to avoid another `mas_common_robotics`. `mas_utils`? `mas_tools`?

argenos, 2019-05-26:
I’m in favor of that. Any preference on the name?

alex-mitrevski, 2019-05-26:
I have a small preference for `mas_tools`, but only because I'm biased by `black-box-tools`. :)

minhnh, 2019-05-26:
Either way is fine for me. I would also want to migrate some of the utilities from [`dataset_interface` repo](https://github.com/minhnh/dataset_interface/blob/feature/data_augmentation/utils.py) there for reuse.

argenos, 2019-05-26:
Here it goes [mas_tools](https://github.com/b-it-bots/mas_tools)

minhnh, 2019-05-26:
should it be a ROS package? I suppose there's the possibility for it to contain both C++ and Python.

alex-mitrevski, 2019-05-26:
I'd say yes, mostly because ROS deals with package paths nicely.

minhnh, 2019-05-27:
you can now access the function with `from mas_tools.ros_utils import get_package_path`

argenos, 2019-05-31:
@ErickKramer do you have an estimated date on when you could push the fixes mentioned here? For now this seems to be one of the blocking issues for #168

ErickKramer, 2019-05-31:
Yes, I'll get it done today.

ErickKramer, 2019-05-31:
> The rasa models we are loading are right now hard coded, see [...]

The checkpoint file is automatically generated by the training script of rasa. I am almost certain that it is not needed for the implementation part because it would've complained already when I put the model folder in lucy. 

@argenos should I just move everything the way it currently is inside  [mas_models](https://github.com/b-it-bots/mas_models) under a rasa_nlu/where_is_this folder?

argenos, 2019-05-31:
Yes, but make it a ROS package, I would say: `rasa_nlu_models/common/where_is_this`. Then edit the `mdr_where_is_this` package to load the models from the right place.

argenos, 2019-05-31:
@minhnh can give you further info on how to use the utilities from `mas_tools`

alex-mitrevski, 2019-05-31:
Minh is at lunch at the moment, so I'll answer. Currently `mas_tools` only has a single script inside the package - [`ros_utils`](https://github.com/b-it-bots/mas_tools/blob/master/src/mas_tools/ros_utils.py), which exposes a single function, namely `get_package_path` (the documentation has an example call).

Basically, if `rasa_nlu_models` is a package, then you can find the absolute path of the model directory with the following code snippet:

```
from mas_tools.ros_utils import get_package_path
model_path = get_package_path('rasa_nlu_models', 'common', 'where_is_this')
```

argenos, 2019-06-04:
Is this done? Was there a different PR for this issue other than the one that was closed?

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

argenos, 2019-06-11:
@ErickKramer @alex-mitrevski shall I close this?

alex-mitrevski, 2019-06-12:
Yeah, I think so.

closed 2019-06-12 by argenos