b-it-bots/mas_domestic_robotics #175: Migration of packages to Python 3

argenos, 2019-05-25:


alex-mitrevski, 2019-06-15:
I can't recall if there was a discussion about this, but I think we should treat this issue independent of the Melodic migration for the time being. Because of ROPOD, we cannot (fully) migrate to Melodic now, but most of the code can be migrated to Python 3 anyway since ROS kinetic works fine with Python 3 (with the exception of a few packages that need some fixing, notably `mongodb_store` and `tf2`).

argenos, 2019-06-15:
This is only an issue for the staff, the rest of the team could migrate. We could, of course, try to make our kinetic and melodic branches compatible, I’m just not sure what the current issues are of setting up kinetic with python 3. What about using docker?

alex-mitrevski, 2019-06-18:
> I’m just not sure what the current issues are of setting up kinetic with python 3

As I said, I've only had problems with a few packages (I forgot to mention OpenCV in the above list, which is also problematic); everything else is working fine with Python 3.

> What about using docker?

You mean docker for melodic, right?

argenos, 2019-06-18:
> > I’m just not sure what the current issues are of setting up kinetic with python 3
> 
> As I said, I've only had problems with a few packages (I forgot to mention OpenCV in the above list, which is also problematic); everything else is working fine with Python 3.
> 

Alright, based on that we would need:
- a script to setup Kinetic with python 3 to make it easier for everyone in the team to switch
    - the opencv issues already came up in https://github.com/b-it-bots/mas_perception_libs/issues/5 but maybe @minhnh can update us on the status
- someone to setup the CI to use Kinetic with Python3 (using the script above?)

> > What about using docker?
> 
> You mean docker for melodic, right?

Yes, *we* (the staff) could use docker for melodic. The rest can upgrade or have a bionic partition.

minhnh, 2019-06-18:
In general, OpenCV shouldn't be an issue. The big change was `indigo`->`kinetic` since they went from OpenCV 2 to 3. `kinetic`->`melodic` is just OpenCV 3. I haven't got a chance to add tests for `cv_bridge`, however.

argenos, 2019-06-18:
I reopened #176 and updated it to `kinetic`, and I can take care of it once someone creates the script.

alex-mitrevski, 2019-06-18:
@minhnh I was referring to Python 3 on `kinetic` when I said there are problems with OpenCV; `kinetic` to `melodic` might not be a problem.

argenos, 2019-10-14:
FWIW, on an unrelated test, I ended up having to test something with opencv and python3. `cv_bridge` needs to be built with python3, e.g. see [here](https://stackoverflow.com/questions/49221565/unable-to-use-cv-bridge-with-ros-kinetic-and-python3)