b-it-bots/mas_domestic_robotics #198: Moving the navigation goals to the ontology

alex-mitrevski, 2019-07-05:
Since navigation goals can be considered encyclopedic information that is currently stored in YAML files (e.g. [here](https://github.com/b-it-bots/mas_domestic_robotics/blob/kinetic/mdr_environments/brsu-c069/navigation_goals.yaml)), it makes sense to consider storing the information in the ontology instead in order to centralise the knowledge in one place. An alternative to storing the poses in the ontology is storing them in the [topological map](https://github.com/b-it-bots/topological_map).

This would require minor modifications to the `move_base` action and the navigation goal visualiser, which currently read the goals from the previously mentioned YAML file, as well as the [map annotation tool](https://github.com/b-it-bots/map_annotation_tool), which creates the YAML file.

I don't foresee this to be a lot of work, so when I have some time, I could draft these modifications in order to show in action how I have imagined the workflow.

argenos, 2019-07-08:
I'll also draft the changes for moving them to the topological map

alex-mitrevski, 2020-02-01:
I've started addressing this with https://github.com/b-it-bots/mas_knowledge_base/pull/11

It would be good to resolve https://github.com/b-it-bots/mas_knowledge_base/issues/12 before I proceed however since I wouldn't want to repeat the definitions of classes and properties in multiple ontology files.