b-it-bots/mas_domestic_robotics #202: Migration to melodic

argenos, 2019-07-24:
I've opened a different issue from #175 since I don't want to mix things up. 
Since I had to set up 18.04 on my laptop, I gave it a try with our repos. One thing I noticed, is that it seems there are no packages in bionic for the HSR. Can someone else double-check?

I have an e-mail where they say they support 19.04, but nothing for 18.04. If that's the case, melodic may be out of the question.

minhnh, 2019-07-24:
Even the update manual for 19.04 seems to work with kinetic packages, so I guess melodic is not yet supported by Toyota. It may make sense to ask them directly what the timeline for this upgrade is.

moriarty, 2019-07-24:
It’s possible to write Kinect/Melodic compatibles code- MoveIt is Kinetic/Melodic compatible.

Just build and test master on both versions.

If you want to run Melodic/Kinetic mixed, then you can do that with Docker, as long as the msgs haven’t changed between versions, and most the standard msgs haven’t changed.

moriarty, 2019-07-24:
19.04 isn’t a targeted version of Ubuntu for any ROS release.

http://www.ros.org/reps/rep-0003.html#kinetic-kame-may-2016-may-2021

The decision was made to only target LTS versions of Ubuntu, and ROS Noetic will be the last version of ROS1. After N-turtle everyone should switch to ROS2.

18.04 is LTS, and ROS1 Melodic / ROS2 Bouncy are both LTS.

Supporting 19.04 and Kinetic is very strange.

argenos, 2019-08-06:
Now that I'm back, I had a bit of time to look a bit more carefully. The 19.04 refers to **their** package version numbers, not the OS. As far as I could tell, they are still only supporting `kinetic`. I assume no one touched base with Toyota, so I've sent an email asking just in case.

alex-mitrevski, 2020-02-01:
I'll close this since it's not high on our priority list anymore.

closed 2020-02-01 by alex-mitrevski